OpenRoberta/robertalab #793: take a tour has erratic behaviour in simulation part at low network bandwidths

akshat-khare, 2018-03-16:
I operated the site from an area with slightly less network availability and there were UI related bugs in the simple tutorial parts. My screen size is 15.5" or 1536 x 864 resolution. But it is not happening because of screen size.
The point is: when I operate the site via my institute wifi, which is quite fast .i.e., 3 Mbps but when I operate via my sim card hotspot which give around 200 Kbps, the bug occurs. I know its kinda weird but that is what is happening. For example, this time when I run the tutorial and hover over the play button, it doesn't turn green and change cursor, what it should have done. Moreover, the arrow is also pointing wrong. I can safely deduct that low network speed are causing this issue.
[image]
[image]
I am using google chrome.

rbudde, 2018-05-02:
This issue is hard to test. Using "slow 3G" the tour was almost ok. But problems remain:
- the first link doesn't point to the close button
- resizing the window when the tour is active destroys the layout
- double clicking the "sim" button destroys the whole layout.
We should decide in the team how to achieve more quality.
Please COMMENT on this ticket!

bjost2s, 2018-05-03:
if I use slow 3G with the old version, I got  similar behavior like mentioned in the bug report. With this fix, opening the simulation is triggered "after" the rest call for getting simulation data was answered. 

Resizing while in the tour is not supported by the tour library, also the tour is not available for small screens in general, because it is impossible to layout the content on small screens.