OpenRoberta/robertalab #795: modal-dialog not closing when head-navigation is changed

akshat-khare, 2018-03-16:
While working on simulations, with sensors or other modals switched on, when I change the navigation to gallery, the modals are still displaying. They should be toggled off, whenever the navigation is changed.
[image]
to
[image]

lccopp, 2018-03-19:
Not only in this situation. If you change the page and click one function it will not redirecting back to the original page.like this
[image] I found these when test the bug you mention last time.haha

akshat-khare, 2018-03-19:
Finding bugs while debugging, nice going man