OpenRoberta/robertalab #797: Lists, For-Each and Code Generation

rbudde, 2018-03-19:
related to: #783 #779

**This ticket is to initialize a discussion, not to start implementation!**

- we support lists and for-each over lists
- for lists user expect the usual list operations: add, remove, get

- code generator for c++, java, python are straightforward
- std::list is used for C++. We have to think about memory leaks when we use std::string
- for C-level code generator we have to decide whether we
  - need a kind of small memory management for strings and lists.
  - allow extending and shrinking of lists
  
In any case we have to rewrite the code generators
- because allocation and management of lists is wrong
- assignment and copying (at least for C-level) need more care
- the for-each should be not integrated into binary expressions
- should work for C-level languages

Please comment!

bjost2s, 2018-03-19:
we use std::list only in develop, on master we use array in c++!!!
we should immediately switch back to array, otherwise current programs will not be compilable anymore:
[image]

rbudde, 2018-03-19:
You're right, but please add your comment to an implementation ticket (or write a new one). This ticket is for **discussing this topic**, which may lead later to an implementation solving all the problems mentioned above.

bjost2s, 2018-03-19:
Ok, I just wanted to clarify, that we use arrays for C++ implementations ;-), see #735

VinArt, 2018-03-23:
Reverted: https://github.com/OpenRoberta/robertalab/commit/3edde07628bb4f692545cecf5c7dc389041d845d

rbudde, 2018-04-13:
b.t.w.: the actual implementation allows to access elements without checking **out-of-bound** indices. Both read and write.

boonto, 2018-09-07:
**Definition of done**
 - a concept for discussion in a meeting
 - for this concept extend the [wiki page](https://github.com/OpenRoberta/robertalab/wiki/Lists-vs-arrays-problem)

VinArt, 2018-09-21:
So I tried doing something for calliope (c++ codegen) in regard of lists.
[image]
Basically I used std::list container and implemented the set and get operations using iterators. There are of course issue with OutOfBounds and probably others, but this sketch could be an example.

rbudde, 2018-09-24:
Artem+Reinhard: for calliope + c++ it looks ok. Writing of putListElementByIndex would be fine.

Arduino is the only robot which needs to be considered. Port of STL is a good solution, if not too resource-hungry.

boonto, 2018-09-27:
We should also look into the comparison operation for lists.

VinArt, 2018-09-27:
[Here](https://en.cppreference.com/w/cpp/container/list/operator_cmp) it says that comparison operators for std::list are overloaded, so that equal/not equal:
> Checks if the contents of lhs and rhs are equal, that is, they have the same number of elements and each element in lhs compares equal with the element in rhs at the same position.

and less/greater and the company:
>Compares the contents of lhs and rhs lexicographically. The comparison is performed by a function equivalent to std::lexicographical_compare.

for [java lists](https://docs.oracle.com/javase/8/docs/api/java/util/List.html#equals-java.lang.Object-):
>Compares the specified object with this list for equality. Returns true if and only if the specified object is also a list, both lists have the same size, and all corresponding pairs of elements in the two lists are equal. (Two elements e1 and e2 are equal if (e1==null ? e2==null : e1.equals(e2)).) In other words, two lists are defined to be equal if they contain the same elements in the same order. This definition ensures that the equals method works properly across different implementations of the List interface.

for python there is a topic on [stackoverflow](https://stackoverflow.com/questions/9623114/check-if-two-unordered-lists-are-equal)

boonto, 2018-09-28:
We compiled all information in the [Lists wiki page](https://github.com/OpenRoberta/robertalab/wiki/Lists).

closed 2018-09-28 by boonto