OpenRoberta/robertalab #798: Extend Nepo language for parallel programming

bjost2s, 2018-03-20:
The idea is to offer additional start-blocks for all systems that are capapble to execute threads / task in parallel.

ensonic, 2018-03-20:
The main issue that needs to be considered is accessing the same resource from multiple 'threads', e.g. one threads does `[turn left 90°]` and the other thread does `[turn right 45°]`. Classically the one executed later will overwrite the previous call and win. This is hard to debug.

bjost2s, 2018-03-20:
Teacher who were asking for this feature told me, that it is ablolutely fine, that users take care about this problem and learn how to use it. Also e.g. Lego Software Ev3 offers this for a long time. The main usecase for this is:
- one task is doing motor control 
- another one in parallel is playing music or listening to sensors

rbudde, 2018-09-07:
the first step should be to write a concept about which kind of parallel execution we want to have and check the possibilities to implement this for different robots. Parallel execution is a lot about communication between parallel executing unit. Furthermore it must be discussed whether associating parallel units with conditions (events, timer) has to be supported or not.

boonto, 2018-09-07:
**Definition of done**
 - a concept for discussion in a meeting
 - for this concept extend the wiki by a new entry