OpenRoberta/robertalab #800: Arduino break/continue loop in wait until block is not working as expected

bjost2s, 2018-03-20:
seems that working cpp implementation in the robot project is overwritten
[image]

bjost2s, 2018-03-20:
@cvejoski could you please work on it?

rishabhdeepsingh, 2018-03-21:
@bjost2s I find a Solution we need to return from the Function as break will only break the last loop and you are using two infinite loops. So returning from the function will solve the issue.
I will send a PR Regarding this.

cvejoski, 2018-03-22:
@bjost2s yes I am on it

elhele, 2018-04-04:
Not resolved. When uploading program, it returns the following error:
"/tmp/userProjects/4cNN_Qa-_7t3/NEPOprog/src/NEPOprog.ino: In function 'void loop()':
/tmp/userProjects/4cNN_Qa-_7t3/NEPOprog/src/NEPOprog.ino:34:5: error: expected primary-expression before '}' token
     }break_loop2:
     ^
exit status 1"

closed 2018-04-05 by bjost2s