OpenRoberta/robertalab #802: Robot can leave simulation arena

VinArt, 2018-03-21:
It is possible to drive the robot outside of the simulation arena under certain conditions, like putting it on the edge/hazard line

[image]

Note Robot X coordinate

rishabhdeepsingh, 2018-03-21:
@VinArt  Can u tell me how did u make the robot go outside the Simulation Area?

bjost2s, 2018-04-12:
This is a special realtime simulation. The robot cannot go out of the room when the door is closed, but if you carry it out of the room, it is out of the room. Definitely the robot cannot go outside the walls by itself. I think that is sufficient.
If you don't agree, please provide some ideas where to place it when it is set outside via the mouse.

closed 2018-05-07 by bjost2s