OpenRoberta/robertalab #803: Calliope is prime and is whole not working

boonto, 2018-03-21:
[image]
Both functions of the math prop block generate uncompilable code.

elhele, 2018-04-04:
Still doesn't work for me on Calliope 2016:
[image]
Calliope 2017 seems to be fine.

closed 2018-04-05 by bjost2s

Fr4n2, 2018-06-29:
Version 2.7.1
the "prime"-block is not properly working.

Works in simulationmode
but cannot download to calliope
[image]

"iswhole"-block is working