OpenRoberta/robertalab #804: Nao communication interface

bjost2s, 2018-03-23:
Currently Nao communicates via the Nao program with the lab server, but Nao is capable to communicate directly to the lab server similar to how EV3 is doing it via Wifi.

As a user I would like to activate the communication between Nao and the lab server via an app from the Nao's app store. Once one ordered the Open Roberta app it will be installed on Nao, so that Nao connects to the server automatically. 
* Nao (the interface) creates a token (parts of the mac adress) and sends it to the lab server
* the user can ask Nao to speak out the token via an event (to discuss with Softbank)
* the user provides the token in the GUI to finish the connection

bjost2s, 2018-03-23:
@VinArt please attach here the sequence diagram, to show the interactions between server, GUI and Nao

VinArt, 2018-03-23:
Also a diagram that roughly shows how the client on NAO works:
[image]

closed 2018-12-20 by VinArt