OpenRoberta/robertalab #812: Moving Sim-Windows strange behavior

Thorlogy, 2018-03-26:
Running following program and then reduce or maximise the simulation window  

[image]

leeds to following strange behavior:

[image]

System: Win10 
Browser: Firefox 52.7.2. (64-bit)

Thorlogy, 2018-04-05:
On the real calliope - four LEDs (2,3,4,5,) are turned on (as expected)
When I change the program to following:
[image]

The strange behavior still "exists". Nothing happened in the simulation. Is this still a hint that something is wrong with the program?

In general i think this is not the best way to "show" the user, that the program is wrong.

bjost2s, 2018-04-05:
x and y in "Setze LED" can hold expression or variables, which will change at runtime. Unfortunately there is no way to check this before the program is executed.
your program works different for me (and as expected):
[image]

Thorlogy, 2018-04-09:
i ment - if i change the program from 10 to 5 - the result in the simulation is the same

bjost2s, 2018-04-15:
I will also set the default values for the for loop for calliope from  0 to < 5 step 1

bjost2s, 2018-04-15:
Please test it, have a look at the console for array out of bound errors.

rbudde, 2018-05-02:
the "count with i" block and the "wait until" block are disabled. It's impossible to test.

Issue reopened.