OpenRoberta/robertalab #816: NXT colour sensor "RGB" mode does not work

boonto, 2018-03-26:
The NXT colour sensor does not support an RGB mode.

rbudde, 2018-05-02:
Not clear what to check. Description of the issue to short. Commits give no hints about what to check.

boonto, 2018-05-02:
The basic NXT colour sensor has no method to return an array with RGB values, which was required by the RGB mode. Using the mode led to errors. Because the mode is not supported I removed it from the block, which simply inherited the EV3 colour sensor's modes. I will add more information to the issue and commits in the future.

rbudde, 2018-05-07:
I understand the issue and your solution. Everything seems to work fine.
But I don't find a test sequence to see the difference between the actual and the next version of OpenRoberta. How do I select the RGB-mode in the actual version?

boonto, 2018-05-07:
I just noticed the issue was already fixed in 2.6.0, it shouldn't be on the 2.7.0 project at all.

rbudde, 2018-05-07:
:-)

closed 2018-05-07 by rbudde