OpenRoberta/robertalab #818: Have FAILURE when compile

MRNIU, 2018-03-26:
mvn -v
Apache Maven 3.5.2 (138edd61fd100ec658bfa2d307c43b76940a5d7d; 2017-10-18T15:58:13+08:00)
Maven home: /usr/local/Cellar/maven/3.5.2/libexec
Java version: 9, vendor: Oracle Corporation
Java home: /Library/Java/JavaVirtualMachines/jdk-9.jdk/Contents/Home
Default locale: zh_CN, platform encoding: UTF-8
OS name: "mac os x", version: "10.13.3", arch: "x86_64", family: "mac"

Log:

[INFO] 2 warnings
[INFO] -------------------------------------------------------------
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR :
[INFO] -------------------------------------------------------------
[ERROR] /Users/***/Documents/GSoC2018/OpenRoberta/robertalab-master/OpenRobertaParent/OpenRobertaRobot/src/test/java/de/fhg/iais/roberta/util/test/JavaBeanTester.java:[275,37] 无法访问com.sun.beans.introspect.PropertyInfo
  找不到com.sun.beans.introspect.PropertyInfo的类文件
[INFO] 1 error
[INFO] -------------------------------------------------------------
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] RobertaParent ...................................... SUCCESS [13:49 min]
[INFO] OpenRobertaRobot ................................... FAILURE [07:37 min]
[INFO] RobotEV3 ........................................... SKIPPED
[INFO] RobotNXT ........................................... SKIPPED
[INFO] RobotArdu .......................................... SKIPPED
[INFO] RobotNAO ........................................... SKIPPED
[INFO] RobotMbed .......................................... SKIPPED
[INFO] OpenRobertaServer .................................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 21:27 min
[INFO] Finished at: 2018-03-27T03:11:08+08:00
[INFO] Final Memory: 41M/136M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:testCompile (default-testCompile) on project OpenRobertaRobot: Compilation failure
[ERROR] /Users/***/Documents/GSoC2018/OpenRoberta/robertalab-master/OpenRobertaParent/OpenRobertaRobot/src/test/java/de/fhg/iais/roberta/util/test/JavaBeanTester.java:[275,37] 无法访问com.sun.beans.introspect.PropertyInfo
[ERROR]   找不到com.sun.beans.introspect.PropertyInfo的类文件
[ERROR]
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :OpenRobertaRobot

ensonic, 2018-03-26:
which branch is this? develop? 
The ci thinks the build is green: https://travis-ci.org/OpenRoberta/robertalab
I just built locally (mvn 3.3.9 and java-8) and build/test passes.

MRNIU, 2018-03-27:
I was use both the master and develop, both mvn 3.3 and 3.5, but the same error.

cvejoski, 2018-03-27:
You are using Java 9, at the moment you cannot build with Java 9

MRNIU, 2018-03-27:
@cvejoski 
I changed to Java 8, but 

```
java -version
java version "1.8.0_162"
Java(TM) SE Runtime Environment (build 1.8.0_162-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.162-b12, mixed mode)
```
INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] RobertaParent ...................................... SUCCESS [  0.901 s]
[INFO] OpenRobertaRobot ................................... FAILURE [01:44 min]
[INFO] RobotEV3 ........................................... SKIPPED
[INFO] RobotNXT ........................................... SKIPPED
[INFO] RobotArdu .......................................... SKIPPED
[INFO] RobotNAO ........................................... SKIPPED
[INFO] RobotMbed .......................................... SKIPPED
[INFO] OpenRobertaServer .................................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 01:45 min
[INFO] Finished at: 2018-03-27T19:28:02+08:00
[INFO] Final Memory: 35M/568M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.17:test (default-test) on project OpenRobertaRobot: Execution default-test of goal org.apache.maven.plugins:maven-surefire-plugin:2.17:test failed: Plugin org.apache.maven.plugins:maven-surefire-plugin:2.17 or one of its dependencies could not be resolved: Could not transfer artifact org.apache.maven.surefire:common-junit48:jar:2.17 from/to central (https://repo.maven.apache.org/maven2): Connect to repo.maven.apache.org:443 [repo.maven.apache.org/151.101.196.215] failed: Operation timed out (Connection timed out) -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/PluginResolutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :OpenRobertaRobot

cvejoski, 2018-03-28:
according to the error log from the last build you had connection problems:

> [ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.17:test (default-test) on project OpenRobertaRobot: Execution default-test of goal [...]

MRNIU, 2018-03-30:
Okay, maybe I can find another way to fix it.. Thank you!

closed 2018-03-30 by MRNIU