OpenRoberta/robertalab #819: Will be stuck when opening a webpage

MRNIU, 2018-03-26:
[robertalab's web site](https://lab.open-roberta.org), every time I open it will display a window "Attention" and I can't close it, so I have to Refresh the page. My Web browser is Safair.

rishabhdeepsingh, 2018-03-26:
Can u share the Screenshot of it?

VinArt, 2018-03-26:
Could you also traceroute till lab.open-roberta.org? it might be timing issues.

MRNIU, 2018-03-27:
The Screenshot @rishabhdeepsingh 
[image] 

Still not working, but this page in Chrome were all right.
@VinArt 

Thanks you.

VinArt, 2018-03-27:
I am not sure what I am looking at, but am I correct to see pings up to 0.8 seconds?

MRNIU, 2018-03-27:
It's a successful  example
[image]

And it's a failing
[image]

Their request is different, is this the cause of the problem?

Now that I can visit normally, maybe there is a reason for the time.

closed 2019-01-15 by MRNIU