OpenRoberta/robertalab #820: Web browser going slow when clicking on the code tag

MRNIU, 2018-03-26:
I noticed that the code tag is very slow when I click on it, and make my computer stuck sometime, maybe we can do some optimization here.

ensonic, 2018-03-26:
Do you see any warning in the developer console? Which browser/os are you seeing this on?

MRNIU, 2018-03-27:
Thanks. This is the Screenshot, and it's all right in Chrome.
my info: macOS High Sierea, Safari

[image]

bjost2s, 2018-03-28:
We know that many people from China have problems to access the Open Roberta Lab. The messages in the console are not reproducable for us, unfortunately.
Maybe it is a problem with High Sierra, I will try to find a device next week to test this in detail.

MRNIU, 2018-03-30:
So unlucky. Thank you for your help.

closed 2018-03-30 by MRNIU