OpenRoberta/robertalab #821: Bluetooth communication is not working

spidgorny, 2018-03-26:
Tried to test Bluetooth communication between two EV3 robots. I could not make it work. Is this supposed to work reliably?
Master:
[image]
Slave:
[image]
Master beeps and after a ~ minute shows an error message:
[image]
Slave waits for ~5 minutes and errors like this:
[image]
Where are these error codes documented?
Any ideas what am I doing wrong?
Thanks.

spidgorny, 2018-03-26:
Version: EV3 leJOS 0.9.1 on both devices.
Name "ev1" is set in (B)luetooth settings in menu.

Tried to pair devices to each other in advance - it says UNSUCCESSFUL.
What else can I try?

bjost2s, 2018-03-28:
Make sure after renaming the ev3 that you have restarted the systems. 
Is bluetooth enabled on both robots?

spidgorny, 2018-03-28:
BT was enabled - I hope this setting persists after restart because I didn't check every time.
I've reboted EV3 several times - it does not help. 
So you say it is working for you? Should pairing of two EV3 to each other be done in advance or this is irrelevant? What does these two error codes mean?

bjost2s, 2018-03-28:
It should work. Could you please test it with "EV3" as the name of the brick?. I'm not sure what setting the name in the Bluetooth settings means. So we tested it with renaming the robot via the system, so that the new name appears in the first line of the menu.
Could you also try this program for the Slave?
[image]

spidgorny, 2018-03-30:
Thank you. I've tried rebooting. I've tried your program as slave. Bluetooth is ON and visibility is ON. Tried connecting to "EV3". The error on master is now different "Could not connect in given time". Slave is just waiting with empty screen. This is confusing. Anything else I can try?
Do you have any way to enable debug mode or additional logging? Should I try the latest "master" branch commit? Are there any fixes to BT since leJOS 0.9.1?
Thanks.

bjost2s, 2018-04-04:
Please revise your master program as well. The problem is, that you are trying to connect to the other EV3 in the loop, but EV3 can only hold and use one connection. 
I have attached two programs, master and slave, that work for us.
slave:
[image]
master:
[image]
We will try to make it more clear in the help section for the blocks.

spidgorny, 2018-04-09:
It works with a single connection. (Or maybe after firmware update). Thank you for your help.
"Slave" has to be the actual name of the brick. It's "ev1" in my case.

bjost2s, 2018-04-12:
Thanks for your feedback

closed 2018-04-12 by bjost2s