OpenRoberta/robertalab #823: Password reset bug: Email case sensitive

Dittse, 2018-03-28:
If the entered email address differs in upper and lower case from the one stored in the account, it will not be recognized.

[image]

[image]

bjost2s, 2018-03-28:
Well, the hint "maybe you have spelled it wrong" is correct, or? So your intention is to change the user given email adress to be lowercase. This is not a bug. We have stored the email in the database as it is. I personally think this makes everything more insecure, but let's discuss this and also how we would like to handle already stored email adresses.

Dittse, 2018-08-15:
I just think that it confuses the user if nowhere else the upper and lower case of the email address is important but here it is.

rbudde, 2019-06-26:
lets store the email address as the user typed it and use equalsIgnoreCase when comparing