OpenRoberta/robertalab #852: rad vs degree

rbudde, 2018-04-13:
we have to decide, whether we use rad (pi/2) or degree (90°). This has to be consistent for all robots. Degrees seems to be the best choice (or?).

With Calliope, we have 2 **errors**:
- the simulation expects rad, the robot degrees
- asin(1) on the robot returns 89.1 instead of 90.0

ensonic, 2018-04-14:
+1 for degree. At least in Germany this is the unit that is used in schools, right?

bjost2s, 2018-04-16:
We carefully have to decide when and how we change something. E.g. for real EV3's some implementations of  trigonometric functions expect and return radians, same for simulation.
When switching to degree, the users programs will fail.

bjost2s, 2018-05-07:
Seems that we should start to work on this. 
Let's decide to use degree and only degree for NEPO, please comment if you can't agree on this.

ilincasuzana, 2018-06-25:
+1 degree