OpenRoberta/robertalab #853: make graphical code better readable by non-programmers by using correct grammar

Superwallah, 2018-04-15:
Some text blocks are not named in a way that they create a readable sentence when connecting those blocks.

1. sensor data objects (green with blue puzzle nose on their left side) like degree celsius of thermometer or brightness of light sensor should not start in the naming with "get" as they are built into the code as an grammar object, not as a relative clause including verb. 
Example: I want to check for the compass pointing North.
"If get angle° compass sensor = 0" is not understandable and does not seem "correctly programmed" for a kid. Why not can the block "degrees of compass"? Then it would be "If degrees of compass = 0" and "set k to degrees of compass" instead of "set k to get value° compass sensor".
I would advice to following changes:

- get xxx gesture ==> Calliope is moved xxx
- get angle° compass sensor ==> angle of compass (or "degrees of angle of compass")
- get sound% microphone ==> loudness from microphone (or "% of loudness from microphone")
- get value ms timer 1 ==> milliseconds of timer 1 (or "milliseconds of timer" as there is only 1 timer. And: kids don't know what "ms" is, "milliseconds" may give them a feeling.)
- get value° temperature sensor ==> degrees Celsius from thermometer
- get value signal strength ==> signal strength
- get value% light sensor ==> brightness  (or "% of brightness of light sensor")
- get analog pin 1 ==> analog pin 1
- get angle° gyroscope x ==> angle of x gyroscope (or "degrees of angle of x gyroscope")
- get value milli-g accelerometer x ==> acceleration in milli-g

2. It is possible to embedded blocks into others instead of appending at the end. So why not use it to allow the natural order of words.
I would advice to call it "wait xxx milliseconds" instead of "wait ms xxx".

3. "repeat 10 times do" are two verbs in one sentence. "repeat 10 times:" (colon at the end) will be clear enough, "do" as a word is not needed. Or use "repeat 10 times this:"
And: "repeat indefinitely this:" instead of "repeat indefinitely do".

4. "integer" is no term a newby understand. Why not call it "random whole number between xx and yyy" instead of "random integer from xx to yyy" ?

Shall I change it myself and offer a pull request? As there are 0 pull requests for this project, I am not sure if you would look at this. So I first wanted to discuss this here as an issue.

bjost2s, 2018-04-15:
Thank you very much for your proposal. We will get back to you asap.
Please don't forget, that our blocks are NEPO blocks, not specific Calliope blocks, so renaming them to something with Calliope inside might cause problems with other systems.

Superwallah, 2018-04-15:
Okay, I will update my former post from
get xxx gesture ==> Calliope is moved xxx
to
get xxx gesture ==> is moved xxx

bjost2s, 2018-04-16:
We have designed NEPO for kids and for other newbies in programming. The lab is online since more than 3 years and the feedback is almost exclusively positive. People like to learn something from real programming, using getters and setters, compare it with the source code .... Also our youngest users, age 8+ have no problems to read it.

We do not see any need to change the blocks like in your proposal. Also common blocks that we use are taken from blockly, they did a log of research in this.

Nevertheless, thank you very much for your interest and offer to help us.

Superwallah, 2018-11-01:
Luckily, Makecode from Microsoft is following my suggestions in a better way. Programms written in MakeCode is more human-readable than those in Open Roberta. 

I do not understand why you conclude from "3 years of almost exclusively positive feedback" that improvement is not appreciated. The success of Ruby e.g. is due to its natural way to read and write.
Only because the world is full of Perl enthusiasts does not mean that there is room to create something that is closer to human interaction. Over decades people were happy to interact via a keyboard with a computer. Then Amazon proposes to talk to your computer and the whole world starts buying an Alexa - also those who are very happy keyboard users coding in Perl.

Can you please refer to the research results (web link) that you mentioned from blockly?

ensonic, 2018-11-04:
@Superwallah, please also understand that we have an existing userbase and this is quite a breaking change. I can definitely see your point and it might be a good design if we would just have started.