OpenRoberta/robertalab #856: "create programm link" creates always a link to lab.open-roberta.org

corvus-albus, 2018-04-17:
For a competion nearby I build a openrobertaserver on a raspberry pi. With the extended howto everything worked well an the setup was rather easy. 
The pupils are used to save the program via "create program link". In the case the server breakes down, this is the only way to save the program. The export function needs the connection to the server. 
There is just one point irritating. The program link always point to https://lab.open-roberta.org even if it is createted on a local server. To use the link on the local server again you have to edit it and exchange the https://lab.open-roberta.org with http://localserver. 
I would expect that the program link reflects the server that is used.
Cheers corvus-albus

bjost2s, 2018-04-18:
we could easily change the link to point to the origin server, but than people cannot easily use this link publish fancy programs in the internet, e.g. link from their webpages, pdf ...

Wouldn't it be easier, that people export programs via the export functions and share this files in the intranet?

corvus-albus, 2018-04-18:
It would be possible to save as a xml file. But I have the situation yesterday in mind. I coached six teams at a small competition in Göttingen and the raspberry freezed once or twice. And in this situation the only way to save the recent changes of the program is to create a link. You can't download a xml-file if the server does not respond but you can still create a link. 
As the teammemebers are at the age of 11 and were a little bit stressed, I had to copy the link to an editor, edit the head of the link and copy it back to the browser as the server restarted. It would have been much easier and the teams could have solved the situation on their own, if the link would reflect the localserver. 
This was my experience and I have to admit it is a very rare and special situation. On the other hand I think that there won't be so much people that run their own server. So only a few people would be effected by changing this link feature. And mostly this should be the one that can deal with it, if they want to share their programs via link. 
This are just my thoughts, you have much more contact to the users of openroberta and more knowledge of how people use it. So in the end it is up to you to decide. And if it stays like it is, I will teach the club members how to deal with it in regard of local servers.
Cheers corvus-albus

NestorLarroca, 2018-05-24:
Hello. You have an answer **_Unable to connect to NXT on Ubuntu 16.04 #11?_**

closed 2019-04-18 by bjost2s