OpenRoberta/robertalab #858: Difficulty in following instructions in the wiki concerning Ecllipse setup

akshat-khare, 2018-04-19:
It is quite cumbersome for one who has started to use Eclipse to directly infer from the actions mentioned in Eclipse setup to configure the 'save actions'. I have added information concerning, which all fields are to be selected  would help one in doing setup swiftly.It is in a repo made by cloning the wiki of the robertalab repo. Here is the [repository link](https://github.com/akshat-khare/OpenRobertaWikiLocal) and here is the [updated Eclipse-setup.md file](https://github.com/akshat-khare/OpenRobertaWikiLocal/blob/master/Eclipse-setup.md). Since I am facing difficulty in generating pull request to the wiki, I request your help for it. Or you can directly commit the changes yourself by looking at the [commit](https://github.com/akshat-khare/OpenRobertaWikiLocal/commit/afe2b6c2e178583be3b6d2b39a4b140cf599ddbc) and doing the necessay changes. Thanks

ensonic, 2018-04-27:
One alternative would be to use a java code formatter that runs as a git-hook. We could use e.g. 
https://github.com/google/google-java-format
and configure s a maven target. WDYT?

ensonic, 2018-04-27:
Yes, as a sample in a different project I have this script cehcked into git (e.g. under scripts/):
```bash
!/bin/bash
# git hook to ensure code style
# ln -sf ../../scripts/pre-commit .git/hooks/

result=0

java_files=$(git diff --name-only --staged --diff-filter=ACMRTUXB | grep "\.java$")
if [ -n "$java_files" ]; then
  which >/dev/null google-java-format || (echo "Please install google-java-format"; exit 1)
  diff=$(diff -u <(cat $java_files) <(google-java-format $java_files))
  if [ -n "$diff" ]; then
    echo "$diff"
    echo "To fix, run: google-java-format -i $java_files"
    result=1
  fi
fi

# ... add similar blocks for other languages here

exit $result
```

One can easily find simmilar scripts on the web. The first step is to find a suitable code formatter (for java) though.

closed 2019-04-18 by bjost2s