OpenRoberta/robertalab #861: Program suddenly "not a valid NEPO program" anymore

PeterNerlich, 2018-04-23:
At some point, adding only a few blocks and compiling the Calliope Mini program glitches out the function last worked on to the top left of the canvas next to the `start` block. The compilation seems to work fine, but the workflow is severely interrupted, since the old function is basically inaccessible. Trying to save the project before compilation and re-importing it afterwards (even after reloading) results in the message "»NEPOprog.xml« is not a valid NEPO program and cannot be uploaded!" — The XML itself seems to be fine.

[image]

[XML before and after](https://github.com/OpenRoberta/robertalab/files/1938061/tetris.zip)

PeterNerlich, 2018-04-23:
Firefox 59.0.2 (64-bit) on ElementaryOS 0.4.1 Loki (essentially Ubuntu for these purposes)

bjost2s, 2018-04-24:
I cannot reproduce this error neither on chrome nor on firefox with ubuntu 16.04. I can import tetris1.4.xml and compile it. Afterwords the program looks exactly like before.

Anyone else who can reproduce this error?

PeterNerlich, 2018-04-24:
Yes, 1.4 works. 1.4.1 does not work, despite being almost the same.

boonto, 2018-05-25:
To reproduce with 1.4:
- Copy one of the note functions in the play function, e.g. noteQuarter
- Insert it into the play function, but outside of the if block
[image]
- Open code view or compile the code -> program collapses
**or**
- Export program -> cannot be imported anymore

To fix:
I reduced the program to the minimal point at which it fails:
[image]
Now when opening the code view it collapses.

I rebuilt this simple program with the current lab version (2.7.1) and the same error does not occur.
The only difference is the 'inline="true"' keyword which is missing in the newly built version.

So when removing this keyword from all blocks in the tetris1.4.1.xml it works without problems.
I don't know why this happens when inlining is activated and only with this special program structure, but I guess we removed the keyword a while ago and this shouldn't happen anymore.

rbudde, 2019-06-26:
please re-test