OpenRoberta/robertalab #867: shorten the pairing token

ensonic, 2018-04-27:
Whenever a ev3 disconnects, one will need to re-enter the token. Right now this consists of 8 characters which are numbers + characters (except 01 and IO). That is 32 chars and therefore we have 32^8 token combinations:

| # chars | combinations |
| --: | --: |
| 8 | 1.0995116e+12 |
| 6 | 1.073.741.824 |
| 5 | 33.554.432 |
| 4 | 1.048.576 |

The token needs to be unique for the currently active (ev3) sessions. I believe we could easily reduce the length to 4.

If we are worried about the increase in likelihood to create conflicts we should reconsider #32 .

bjost2s, 2018-04-27:
All robots exept Calliope and Microbit are using tokens.

ensonic, 2018-05-23:
Do we have number of how many active session we have at the same time? I'Ve received more feedback where people complain that it is hard to enter the token.

closed 2019-06-26 by rbudde