OpenRoberta/robertalab #877: NAO turn/walk blocks use radians instead of degrees

VinArt, 2018-05-07:
NAO blocks use degree symbol:
[image]
The tooltip suggests degrees as well:
```
'Turns the robot for number of degrees. 
Only enter positive values and use the dropdown to select the direction. 
It is possible to enter values up to 360 degrees.'
```

whereas the implementation expects radians:
```
def walk(self, x, y, theta):
        rad = math.radians(theta)
        self.motion.moveTo(x / 100.0, y / 100.0, rad)
```
yes, turning is implemented as walking with zero X and Y values.

Also, API description expects radians as well:
```
theta – Rotation around the Z axis in radians [-3.1415 to 3.1415].
```

So a conversion is needed. Otherwise NAO will be turning too much. (1 rad. ~ 57 deg.)

bjost2s, 2018-05-07:
Please check all related actions/sensors for NAO, "walk" might not be the only one

VinArt, 2018-05-23:
My bad, radians converts from degrees to radians.

closed 2018-05-23 by VinArt