OpenRoberta/robertalab #879: Add additional pins to Calliope

boonto, 2018-05-07:
Calliope has two more rows of pins that currently aren't implemented.

- [ ] Blockly
- [ ] Code generation
- [ ] Tests

bjost2s, 2018-05-18:
Please check if Pins 13 - 19 can be added aswell

boonto, 2018-05-18:
C16 to C19 are in now.
I am not really sure about C13, C14 and C15, as they are excluded in the PXT/Makecode editor.

boonto, 2018-05-25:
So I tested all of the pins and their behaviour with a blinking LED via a breadboard:

Pin | No action | Blinking | Also affects
---  | --- | --- | ---
C00 | nothing | blinking | P0
C01 | nothing | blinking | P1
C02 | nothing | blinking | P2
C03 | nothing | blinking | P3
C04 | glowing | blinking but not as programmed | Some display LEDs
C05 | glowing | blinking but not as programmed | Some display LEDs
C06 | glowing | blinking but not as programmed | Some display LEDs
C07 | glowing | blinking but not as programmed | Some display LEDs
C08 | glowing | blinking but not as programmed | Some display LEDs
C09 | glowing | blinking but not as programmed | Some display LEDs
C10 | glowing | blinking but not as programmed | Some display LEDs
C11 | glowing | blinking but not as programmed | Some display LEDs
C12 | glowing | blinking but not as programmed | Some display LEDs
C13 | weak glowing | blinking but with double the interval | Turns off LEDs
C14 | weak glowing | blinking but with double the interval | Turns off LEDs 
C15 | weak glowing| blinking but with double the interval | Turns off LEDs
C16 | nothing | blinking | right Grove RX
C17 | nothing | blinking | right Grove TX
C18 | nothing | weak blinking | left Grove SDA
C19 | nothing | weak blinking | left Grove SCL

Maybe the differing behaviour has something to do with the pull or additional resistors in the LED pins, I don't really know.
I assume because of the highly varying behaviour of C13, C14 and C15 they were left out of the PXT/Makecode editor.

But they still work (somewhat), so we could add them.

closed 2018-07-17 by bjost2s