OpenRoberta/robertalab #880: Import xml-Program

Thorlogy, 2018-05-08:
The first attempt to import a xml Program (EV3, Calliope,...) does not work. 
The second attempt always works.

Browser: Chrome

elenadorogush, 2018-06-25:
I've just tried to reproduce this, but XML program import for EV3dev worked fine on the first try. My Chrome version (chrome://version) is 67.0.3396.87.

bjost2s, 2018-06-25:
this is already solved with #874 @Thorlogy ,@elenadorogush could you please both test it at https://test.lab.open-roberta.org and give feedback?
Thank you

Thorlogy, 2018-06-25:
works. But often the start-block moves and sometimes the whole program is shiftet like following example:

[image]

Thorlogy, 2018-06-26:
here you go:
Program for Calliope 2017:

[Schere.zip](https://github.com/OpenRoberta/robertalab/files/2135719/Schere.zip)

closed 2019-04-18 by bjost2s