OpenRoberta/robertalab #901: EV3 USB Programm for Mac not working

Thorlogy, 2018-06-01:
After downloading and starting the recommended program https://github.com/OpenRoberta/robertalab-usbprogram/releases 

Following Massage appears:
[image]

Mac OS:
[image]

VinArt, 2018-06-01:
[image]

bjost2s, 2018-06-01:
@Thorlogy could we please have your macbook to make some tests next week?
As you can see above, ours is working.

Thorlogy, 2018-06-01:
Yes - on Monday.

bjost2s, 2018-06-01:
Could you please also copy and paste the content of the file Users/thorstenleimbach/OpenRobertaUSB/OpenRobertaUSB.log

Thorlogy, 2018-06-04:
01/06/2018 10:03:22.399 - [de.fhg.iais.roberta.usb.Main.configureLogger] - [INFO] - Logging to file: /Users/thorstenleimbach/OpenRobertaUSB/OpenRobertaUSB.log
01/06/2018 10:03:24.408 - [de.fhg.iais.roberta.usb.Main$1.getLocals] - [CONFIG] - Language de
01/06/2018 10:03:24.412 - [de.fhg.iais.roberta.connection.EV3USBConnector.<init>] - [CONFIG] - Robot ip 10.0.1.1

closed 2019-04-18 by bjost2s