OpenRoberta/robertalab #910: possible Exploit(xss)

K0IN, 2018-06-11:
There is an error while loading XML saved code you could inject javascript and execute some Malicious Code

So just load this file from Edit->Import program
[poc.zip](https://github.com/OpenRoberta/robertalab/files/2089246/poc.zip)

rbudde, 2018-09-07:
**Context**
We believe, that this kind of injection is only possible when importing manipulated programs. Of course, manipulated programs loaded from the database are affected indirectly, too (thus shared program and the gallery are very dangerous!).

We believe, that this injection is only possible for the program documentation, that can be written in its own tab and then exported together with the program. To be bundled with the blockly-XML, special charcters as '<' are encoded as '&lt;' and putting these '&lt;' into the exported program documentation is the basis of this exploit.

**Solution**
When the program is imported, the server should check the program documentation for legal encodings. For this a white list of allowed encodings has to be prepared. All encodings not found in the white list have to be removed.

1. write a utility class, that gets a string and returns a cleaned string.
2. write tests
3. use this class at server side to clean the program documentation.

**Definition of Done**
add a program to the integration tests, that has '&lt;script&gt;alert(42)&lt;/script&gt;' in the program documentation and check, whether this is removed.

VinArt, 2018-09-20:
I added a test that checks whether scirpt tag is removed by invoking the method I wrote, but I think you meant adding it in a way, so it would be using the REST interface (yet to come).

rbudde, 2019-02-12:
1. the changes in the database work. They change `<br>` to `<br />`, which should be no problem.
2. importing of xml is safe now.

closed 2019-02-12 by rbudde