OpenRoberta/robertalab #915: EV3 Functions can't handle "say"-Blocks

Dittse, 2018-06-13:
[image]

This applies for all combinations of "says"-Blocks and functions

boonto, 2018-06-13:
I just checked: this happens due to the Exceptions thrown by the hal.sayText function, which would normally be caught by the catch all in the run method.
I guess just ignoring the exceptions in the Hal would be possible, retroactively adding throws to userdefined functions does not seem nice.

bjost2s, 2018-06-17:
I cannot reproduce this error:
[image]
Could you please give us more information, please upload the whole program, if possible

bjost2s, 2018-06-20:
@boonto the error message was coming from the code generation, which has no access to hal at that moment. I cannot reproduce this error on lab.open-roberta.org.

boonto, 2018-06-20:
@bjost2s the block has to be inside a function block.

bjost2s, 2018-06-21:
@boonto sorry, now I understand.

closed 2018-07-17 by bjost2s