OpenRoberta/robertalab #920: NXT code generation error

rbudde, 2018-06-18:
writing promitive types to the display invokes c_str() on them, which fails.

```
10:07:24.368 [main] INFO d.f.i.r.j.i.CompilerWorkflowIT - [[[[[ Robot: ardu compile nepo file: /crossCompilerTests/ardu/standardSensors.xml =====
10:07:24.368 [main] INFO d.f.i.r.j.r.all.ClientAdmin - command is: setRobot
10:07:24.368 [main] INFO d.f.i.r.r.RobotCommunicator - token 3_1--b4QMssR is not waited for. Ok.
10:07:24.368 [main] INFO d.f.i.r.j.r.all.ClientAdmin - set robot to ardu
10:07:24.369 [main] INFO d.f.i.r.j.r.all.ClientProgram - command is: compileP, userId is -1
10:07:24.431 [main] INFO d.f.i.r.j.r.all.ClientProgram - compiler workflow started for program prog
10:07:24.434 [main] INFO d.f.i.r.f.AbstractCompilerWorkflow - stored under: /tmp/userProjects/toknTokn/prog/src/prog.ino
/tmp/userProjects/toknTokn/prog/src/prog.ino: In function ‘void loop()’:
/tmp/userProjects/toknTokn/prog/src/prog.ino:53:50: error: request for member ‘c_str’ in ‘bnr.BnrRoberta::colorSensorLight(((byte*)(& colorsLeft)), 1)’, which is of non-class type ‘int’
     one.lcd1(bnr.colorSensorLight(colorsLeft, 1).c_str());
                                                  ^
exit status 1
10:07:25.095 [main] INFO d.f.i.r.f.a.b.CompilerWorkflow - COMPILERWORKFLOW_ERROR_PROGRAM_COMPILE_FAILED
10:07:25.095 [main] INFO d.f.i.r.j.r.all.ClientProgram - compileN terminated with COMPILERWORKFLOW_ERROR_PROGRAM_COMPILE_FAILED
10:07:25.096 [main] ERROR d.f.i.r.j.i.CompilerWorkflowIT - ----- Robot: ardu compile nepo file: /crossCompilerTests/ardu/standardSensors.xml FAILED(response-info) ]]]]]
```

elhele, 2018-06-18:
@rbudde it seems to be Botnroll, not NXT

rbudde, 2018-06-18:
@elhele you are right. It wasn't NXT, but arduino!

bjost2s, 2018-08-27:
Do we have to fix it anyway?