OpenRoberta/robertalab #923: revise openroberta service on the test server

rbudde, 2018-06-19:
we have a service on the test server, thar runs the develop version of OpenRoberta. This service is restarted when a commit in develop is detected. The way, this is done, is not robust. There are timing issues (multiple commits, rebuild + startup too slow?), that let the restart fail. Often the database could not be restarted (port in use, e.g.). If you log in into the test machine and execure a 

> service robertalab.sh restart

everything is fine. But that's no solution :-)

bjost2s, 2018-06-20:
The problem is:
018-06-20 09:30:59,710 [main] session-id=[] user-id=[] robot-name=[] ERROR d.f.iais.roberta.main.ServerStarter - Could not start the server at http://0.0.0.0:1999
java.net.BindException: Die Adresse wird bereits verwendet
from the ora-server.log
looks like the stop server command is not executed successfully when a commit is detected, but when doing service robertalab.sh restart, the same stop routine is executed and then it works!

bjost2s, 2018-06-20:
I added a sleep 10 after the stop command in the update routine. This should not fix the problem, but if so,
we should try to understand the execution of the later commands better.

closed 2019-04-18 by bjost2s