OpenRoberta/robertalab #924: Allow users to start the lab with the desired robot system preselected

bjost2s, 2018-06-20:
This can be achieved by adding dedicated parameter(s) to the url like we do it for the "create program link", e.g. https://lab.open-roberta.org/#loadProgram&&ev3lejosv1&&...
We could agree on something like https://lab.open-roberta.org/#**loadSystem**&&ev3lejosv1.
Functionallity should be added in [menu.controller.js](https://github.com/OpenRoberta/robertalab/blob/develop/OpenRobertaParent/OpenRobertaServer/staticResources/js/app/roberta/controller/menu.controller.js) 
[image]

ensonic, 2018-06-20:
url params are usually in the form:
```
https://lab.open-roberta.org/path?key1=value1&key2=value2&...
```
(see https://en.wikipedia.org/wiki/Query_string).

Do we support any parameters right now? Do we have some docs about it (e.g. a wiki page)?

bjost2s, 2018-06-20:
We dont have a wike page for this, but as you can see in my comment above what we already support.

ensonic, 2018-06-21:
I'd suggest to do `&` and not `&&` and use `key=value`. E.g. `robotSystem=ev3lejosv1`.

alexfh, 2018-06-25:
I'd take a look at this.

elenadorogush, 2018-06-25:
Should this issue be marked as a duplicate of #498?

alexfh, 2018-06-25:
Probably yes, or the other way round. Thanks for fixing this, btw!

commit 8525bc72 (2018-06-27): Support URLs that preselect robot type and skip the startup screen.

fixes #924
closed 2018-07-16 by bjost2s