OpenRoberta/robertalab #928: Make trash bin larger to expand to right hand side of the screen

mmarschall, 2018-06-25:
currently there are a few pixels right of the trash bin, where I can drop a block on the canvas instead of the trash bin.

This causes a lot of usability issues for children - I regularly see blocks piling up on the canvas around the trash bin.

mmarschall, 2018-06-25:
The issue seems to be that as soon as the pointer leaves the canvas, the trashcan closes (no matter whether below or right of the trashcan).

If I drop the block while the pointer off canvas, it still places the block (but on the canvas, not in the trash, as that is closed already).

mmarschall, 2018-06-25:
Main issue is the 4px border on the right hand side, which gets generated by blockly (in my example for a 1440px screen):

<div id="blockly" style="width: 1440px;">
<div id="blocklyDiv" style="... width: 1436px; ..." ...>

bjost2s, 2018-06-25:
I would recommend not to change the design or layout of the bottom button bar, but to check if we can extend the radius where the trash lid opens, so that blocks to be deleted will be fetched by the open lid event. It also seems that the center from the radius is calculated wrong, maybe from former times where the whole bar was smaler (long time ago). It should be in the middle of the upper edge from the trash (without lid). The 4px border is the standard gap between navi icons background for all "squares".

mmarschall, 2018-06-26:
To improve usability of the trashcan I moved it to the toolbox. As blockly deletes every block you drop on the toolbox automatically, only an icon to make this feature discoverable is needed.

The default blockly trashcan cannot easily be expanded and the 4px border on the right should stay.

This solution has two advantages:
1. The drop area is huge now (whole left side of the screen)
2. You "put blocks back where you found them" is an easier mental model to grasp than looking at the lower right border (where action buttons are which have nothing to do with manipulating blocks).

bjost2s, 2018-06-27:
There a couple of reasons, why we cannont accept your pull request. 
- the new trashcan icon has no functionality, so far all icons on the menu or lower toolbar are clickable
- the new trashcan icon hides parts of the toolbox on small screens
- the lower toolbar is not right aligned anymore
- we do not want to change the GUI unless there are important reasons like bugs etc. Kids have no problem to use the trashcan where it is. They like it, when lid is opening and closing ....
I will have a look on the trashcan and try to increase the radius to make it easier to open the lid

bjost2s, 2019-03-11:
Will be improved here #1159

closed 2019-03-11 by bjost2s