OpenRoberta/robertalab #929: Move cookie acceptance dialog to top of screen to avoid it hiding the start button

mmarschall, 2018-06-25:
The current cookie acceptance banner covers the lower part of the screen. 

It covers the "run" button, which is confusing to children who want to get started with their program.

We should move the cookie acceptance banner to the top of the screen so that it doesn't hide any important UI elements.

ilincasuzana, 2018-06-25:
There's menu items at the top and the logo at the right. What about bottom left?

bjost2s, 2018-06-25:
This is afaik intended, because you force the user to read it and to decide.

ilincasuzana, 2018-06-25:
I just committed moving it to the left. Should I revert?

Personally I'm also in favor of making it less invasive but I'm not sure what the official requirements are.

bjost2s, 2018-06-25:
@Thorlogy, could you please comment on this?

Thorlogy, 2018-06-25:
I would suggest placing the cookie hint so that it is clearly visible but does not obscure the run button. I will discuss it internally tomorrow and make an officially agreed proposal.

mmarschall, 2018-06-26:
any guidance on how to place the cookie hint?

Thorlogy, 2018-06-26:
@mmarschall We have decided to leave the cookie notice as it is.

Due to the fact that we work with publishers such as Cornelsen, who rely on the layout to look the same over a longer period of time, all layout adjustments / changes should only be made if they are based on a bug or an absolutely necessary usability decision. 

If the layout differs from the printed version too often or at critical points, the pupils may no longer be able to find their way around. We definitely want to avoid that.

ensonic, 2018-06-30:
Thanks for sharing your thoughts. FYI: this change was made as a result of observing how often kids are puzzled with this notice covering some essential UI elements. Since we use anonymous guest sessions (like incognito mode), this notice comes up every time.

Thorlogy, 2018-07-02:
Thank you. This is an important hint for us. We'll discuss this within our next meeting again.

bjost2s, 2018-07-04:
if we put it on top of the screen, it will hide the menu. The result of a hidden menu is different, but also there kids will not see, that they can use the menu and all the functionality.

mmarschall, 2018-07-05:
What about just making it slimmer?

Thorlogy, 2018-07-05:
Yes, I think it will come to such a solution. 
We'll probably just make the hint a little shorter. Thus the start button would remain visible despite.

philippmaurer, 2018-07-10:
As discussed with parts of the team, we decided to make the cookie disclaimer just a little bit slimmer on large screens, (640px+ or tablet+) so that the cookie disclaimer will no longer hide important UI elements.
On small devices the cookie disclaimer will be moved slightly up, so it is floating on the screen. 

Screenshots:
Large screens: 
[image]

Small screens: 
[image]

This behavior will allow all users to see the relevant UI elements all the time, but will still force them to either accept or refuse the cookie disclaimer to work with the lab on a long run.

bjost2s, 2018-07-10:
For small screens it looks a bit different, because the normal case is that the start popup is visible like here:
[image]
For me it looks a bit unsorted now.

bjost2s, 2018-07-10:
here an example for Iphone 5
[image]

philippmaurer, 2018-07-11:
We decided to move the cookie disclaimer back to the bottom of the screen for small devices, because of the design issues mentioned above.

This will affect devices like smartphones and small phablets.

bjost2s, 2019-03-07:
I will close this issue, because I think it is already resolved and because of #1155

closed 2019-03-07 by bjost2s