OpenRoberta/robertalab #944: button a pressed - and - get pressed button a

Fr4n2, 2018-06-29:
Version 2.7.1
Using a button it is possible to use the block from the sensors library "button A pressed".

But in the precomposed block "wait till button pressed" it is called "get pressed button A". 

Especially working with children this is a bit confusing as there are two different blocks that have the same meaning and are also translated to the same source code. 
I believe the "get pressed button A"-block is a leftover from an older version.

[image]

bjost2s, 2018-07-04:
Hi, no, the get pressed button is a complete different block than the button A pressed button. We tried to find a way to combine all sensor blocks in a single dropdown and this is the result. The block only exists in the wait until block.
You could try to show the kids createing their own wait until like 
[image]

Fr4n2, 2018-07-05:
Hi, 
thank you very much for your answer.

Can you go deeper in explanation where the difference is? Just to understand.
Thank you very much.

[image]

closed 2019-04-18 by bjost2s