OpenRoberta/robertalab #951: Improve data logging for statistics

bjost2s, 2018-07-12:
time - errors + Errorcoded in
time - system - run sim - logged in / not logged in
time - system - login
time - system - logout
time - system - view gallery
time - system - load program from gallery
time - system - save / load program
time - system - sim background uploaded
time - system - browser/system/screensize... once
time - start session
time - end session

boonto, 2018-07-12:
@bjost2s @rbudde what do you think about structuring the logs in JSON format?

rbudde, 2018-07-12:
I prefer JSON over other formats, because:
1. the client generates JSON already
2. the JSON-toString-method does all the necessary quotations of " etc. This is much more complex with other formats (e.g. csv)

boonto, 2018-07-13:
Okay, I will restructure the logging to JSON.

closed 2018-07-17 by bjost2s