OpenRoberta/robertalab #959: WeDo Robot Configuration Missing Lable

Thorlogy, 2018-07-19:
In the RobotConfiguration two lables are missing:
[image]

I would recommend to name this 

English: "Connector" 
German: "Anschluss" 

It should not caled "Port" because thats the lable for the actors and sensors:
[image]

Thorlogy, 2018-07-19:
Additional note: The lable which belongs to wedo should called "Hub"

bjost2s, 2018-07-23:
this was already fixed in hotfix2.8.1 before you created this issue. Please close

Thorlogy, 2018-07-23:
Sorry the issue still exist

[image]

closed 2018-09-03 by bjost2s