OpenRoberta/robertalab #962: WeDo blank screen Robot Configuration

Thorlogy, 2018-07-19:
After changing from "Progam" to "Robot Configuration" the screen was blank. 

[image]

Beforehand i switched the ports where the infrared sensor was connected (direktly at the hub)

System: Android 8.1.0
Browser: Chrome 76.0.3396.87
Wedo 2.0

rbudde, 2018-09-03:
see issue #960 : this seems to be another variant of the annoying problem with Chrome: if view is set to 110% e.g., the width information is delivered wrong by jquery (some other zoom level show the same behavior, others don't). This will hide tabs. If the window is resized (a bit), the tab re-appears.

working now

bjost2s, 2018-09-03:
some improvement available, please test

closed 2018-12-20 by VinArt