OpenRoberta/robertalab #964: Wedo and wifi

Thorlogy, 2018-07-26:
When I use the app, I am notified that I do not have a WLAN connection. 
Why do I need a WLAN connection?

[image]

bjost2s, 2018-08-12:
The Open Roberta App has no inbuild robertalab server. If you want to use the app, you have to access either a local server or lab.open-roberta.org.
AFAIK there is currently no other way for Android mobile devices to connect to a network than via a wifi connection. Thats why the wifi connection is required to use this app.
(Internet access is not required but recommended)

Thorlogy, 2018-08-12:
Sry i do not understand. You mean that Android Apps need WiFi connection to connect with the internet? Or what do you mean with "Network"?

closed 2019-06-26 by rbudde