OpenRoberta/robertalab #967: Lab unable to delete or share saved Programms

Dittse, 2018-07-30:
If you try to delete or share one of your saved Programms, the corresponding window pops up for a very short amount of time and then disappears because the programm is loaded in the workspace.

[image]

Dittse, 2018-08-15:
Now sharing and deleting works fine

rbudde, 2019-02-12:
the problem is solved. But at least the german and the english text are VERY MISLEADING:

The 2.paragraph reads "Wenn du ein geteiltest Programm löscht, löscht du nicht das Programm selbst, sondern das Teilen mit dir.". I deleted one of my programs, which I shared with another user. In this case both the share and the program were deleted.