OpenRoberta/robertalab #970: Calliope C-Code generation produces compiler error, if return value is an array

kunkunuzzi, 2018-08-05:
Any function returning arrays will cause compiling errors. In the generated c++ code, the function definition should be

array<double, N> macheEtwas() {
        return {0, 0, 0};
}

but ora just produces:

array<double,  macheEtwas() {
        return {0, 0, 0};
}

rbudde, 2018-09-03:
`array<double, N> macheEtwas() { return {0, 0, 0}; }` is generated now. But now the error described in ticket #971 applies. The `N` should be a template parameter. Please comment on this!

rbudde, 2019-02-12:
generated code is wrong:

```c++
std::list<double>M> d() {
        return item;
}
```

bjost2s, 2019-02-19:
what is the current state of this issue?
I will remove it from project 2.1.0 "to do" because it is already released

kunkunuzzi, 2019-02-19:
The issue is now existing in the way rbudde describes. But still we have the situation as stated in the title of this issue: "Calliope C-Code generation produces compiler error, if return value is an array", just the error is different.
Example (produces wrong line 21 in C++ code for calliope mini):
(I don't plan to fix this by myself, since I am not involved in this stuff anymore)
 
https://lab.open-roberta.org/#loadProgram&&calliope2017&&Bug970_variante&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22calliope%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22141%22%20y=%2283%22%3E%3Cblock%20type=%22robControls_start%22%20id=%22:64%5B4Bt%7Bdn_qIOZARP/e%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22false%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3E%3C/field%3E%3C/block%3E%3Cblock%20type=%22mbedActions_display_text%22%20id=%22EGPJ~@TAlN:U7StS*Qly%22%20intask=%22true%22%3E%3Cfield%20name=%22TYPE%22%3ETEXT%3C/field%3E%3Cvalue%20name=%22OUT%22%3E%3Cblock%20type=%22robLists_getIndex%22%20id=%22rJeo2dj%7DLwr,r%5DlF-9lL%22%20intask=%22true%22%3E%3Cmutation%20statement=%22false%22%20at=%22true%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EGET%3C/field%3E%3Cfield%20name=%22WHERE%22%3EFROM_START%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22robProcedures_callreturn%22%20id=%22_XZwM3)ROQA%7B#mnm54%5DK%22%20intask=%22true%22%3E%3Cmutation%20name=%22macheEtwas%22%20output_type=%22Array_Number%22%3E%3C/mutation%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22AT%22%3E%3Cblock%20type=%22math_number%22%20id=%22*h(L#1AzSajKD:(!ODW0%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E1%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%22124%22%20y=%22242%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22)I!XBA1RMvHoR0t7?%25H4%22%20intask=%22true%22%3E%3Cmutation%20declare=%22false%22%20return_type=%22Array_Number%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EmacheEtwas%3C/field%3E%3Cfield%20name=%22TYPE%22%3EArray_Number%3C/field%3E%3Crepetitions%3E%3Cvalue%20name=%22RETURN%22%3E%3Cblock%20type=%22robLists_create_with%22%20id=%228m9Q@%7DOT%5EY%60%7CWRFqhV?a%22%20intask=%22true%22%3E%3Cmutation%20items=%223%22%20list_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22LIST_TYPE%22%3ENumber%3C/field%3E%3Cvalue%20name=%22ADD0%22%3E%3Cblock%20type=%22math_number%22%20id=%22oXPna0v,#aM(WXo~ASvk%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E7%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD1%22%3E%3Cblock%20type=%22math_number%22%20id=%22%7CxB4NQ%7CABda%7C4G%605J(LG%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E3%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD2%22%3E%3Cblock%20type=%22math_number%22%20id=%22O/)fwR/+D~7KFg--idlF%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E9%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/repetitions%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/program%3E%3Cconfig%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22calliope%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22138%22%20y=%2288%22%3E%3Cblock%20type=%22mbedBrick_Calliope-Brick%22%20id=%221%22%20intask=%22true%22%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E

closed 2019-06-12 by bjost2s