OpenRoberta/robertalab #973: Calliope: Motor A-Block and Motor B-Block work wrong: they move fast at speed 0 and slower at speed 100!

kunkunuzzi, 2018-08-07:
I tried the abbozza platform, which also translates blockly to C++ using "soundmotor" but it does not show the same error.

bjost2s, 2018-08-12:
Could you please upload your testprogram as xml or screenshot here, so that we can reproduce this bug?
Thank you

bjost2s, 2018-08-27:
Yes, thanks!
It takes some time for us to reproduce this (getting motors, calliopes and oscilloscope together).

VinArt, 2018-08-30:
Hi @kunkunuzzi I have tested your code and noticed one interesting thing: when only 1 motor is set to a certain value, then it goes on a speed, that actually is more than 100% (and it does not matter whether it is 0 or 20, 40, 60 etc) the speed is constant, when I set both motors to some value (having 1 loop instead of 2) then they work as expected, so they are stopped at 0 and then turn faster, when the speed is increased. I am not sure whether it is a hardware issue, or the issue of the underlying library. I will have to look deeper into that. As for now, I can confirm that motors work in unexpected way, when in the 2 motor configuration the speed of only single motor is set to a certain value.

bjost2s, 2018-08-31:
We will offer an additional block that will ask for A **and** B speed to make sure, that the behaviour is correct.

bjost2s, 2018-12-20:
@VinArt please close if this is already released

closed 2019-02-22 by VinArt