OpenRoberta/robertalab #980: Calliope: Cannot login anymore

GerdRuehle, 2018-08-20:
Since I migrated to Ubuntu 18.04.1 LTS I cannot logon anymore. The filed for capturing the password seems to be not open for input. In the log there is an entry

  | jsToAppInterface >invalid webview type caused by: [object Object]
-- | --

That is all I can get you as a hint.

Best Regards
Gerd

bjost2s, 2018-08-23:
Which browser are you using, and is it the same for other browser?
Did you run 17.04 before this or 16.04?
I run Ubuntu 18.04 (fresh installation) and have the same issue but only with Firefox 61.0.1 (64-bit)

bjost2s, 2018-08-27:
I found this here: 
https://support.mozilla.org/en-US/questions/1214123
It seems to be the same issue. I will let you know soon, if the solution (installing firefox from a different repo) will help