OpenRoberta/robertalab #981: Multiple robot simulation : debugging

akshat-khare, 2018-08-23:
### The following tasks were discussed with @bjost2s  and assigned to be done by me regarding multiple simulation feature #976 #977   :
- [ ] Debug ultrasonic sensor to make it detect other robots, ultrasonic sensor not detecting other robots but bump feature is working correctly
- [ ] The multiple simulation works only on the default field, make it working on other fields too
- [ ] Color Values be picked up from the toolbar and random coloring technique should be removed
- [ ] Selection of robots be done via color and not numbering
- [ ] Set the icon of mult sim option to the main menu simulation option 
- [ ] Change the text value of mult sim option to lowercase and other appropriate word
- [ ] Multiple Simulation popup heading be enlarged and font style made similar to the project
- [ ] Search box be indented properly
- [ ] Robot name be removed
- [ ] Css of selection list made similar to that tables used in project
- [ ] Text of Run Simulation and Close changed to Ok and Cancel and rearranged
- [ ] Saved Name be changed to Program name

akshat-khare, 2018-09-03:
The issues were solved and merged in the multiple_robots_simulation branch #990

rbudde, 2018-12-20:
the button multiple robot simulation stopped to work. Why?

bjost2s, 2018-12-20:
Also a lot of unwanted console.log cammands are still in the code

akshat-khare, 2018-12-21:
I'll look into it, it was working before, some commit after my work may have introduced error. I will remove the extra console.log statements also. It might take some time as I am involved in some other engagement. Thanks

bjost2s, 2018-12-29:
As we would like to close develop next week I will revise as much as I can do this weekend.
Thank you again for your valuable contribution

akshat-khare, 2019-01-02:
I compiled code after a long time in my system and I see that the feature is working. The only bug I can see is that when selecting a robot, its code in workspace is not changing. I don't know if that thing is done deliberately or it is a bug that came up during the further change of code. Please clarify @bjost2s .

bjost2s, 2019-01-03:
I have marked this issue as resolved, because I think we can publish it.
There are still some known issues, e.g. the robot's display is not working reliable (only one div for all) which will be fixed in a next release.

akshat-khare, 2019-01-03:
Cool, Thanks for the help

bjost2s, 2019-02-19:
- [ ] Display  and buttons from the robots are still not connected to multiple robots
So we move it to the next release

bjost2s, 2019-02-25:
- [ ] resetting, refreshing and restarting multiple programs in the simulation needs revision

closed 2019-06-12 by bjost2s