OpenRoberta/robertalab #983: Calliope: setPixelValue-Block works wrong, if the brightness is calculated with the addition-math-block

kunkunuzzi, 2018-08-24:
It seems, that similiar as issue #972, a curved bracket is missing somewhere. Only the last argument of this block is affected, since it does a multiplication with the _SET_BRIGHTNESS_MULTIPLIER constant, which gains precedence by C++-Rules. 
Since there might be more blocks affected, it could malke sense to enclose all the the math blocks in curved brackets.

rbudde, 2018-10-01:
@kunkunuzzi  you are right, enclosing everything in parentheses would make it right (and not introduce syntax errors). But as many user look at the generated code (and even ask, whether they can upload modified code), we like to keep the code as readable as possible. Readability decreases when brackets are inserted and not needed. Thus: thank you for your report, we'll try to fix the errors step by step :-)

rbudde, 2018-11-15:
but see #1006 and the same bug in NXT. There is a **general** solution needed.

closed 2018-12-20 by VinArt