OpenRoberta/robertalab #985: BOB3 connected, but transmission not possible

springob, 2018-08-29:
Bob3 programming not possible: programmer will be connected (ORCA), but whenever I click on play-button, the program can't be transmitted (see attached image)'. Problem started with release 2.8.1...

[image]

VinArt, 2018-08-30:
Sorry, we are experiencing problems with the Create Agent with our current deployment, they will be gone in a short period of time when the server will be re-deployed, stay tuned.

VinArt, 2019-02-22:
So with the currently deployed lab I can flash the BOB3, so I will close this. If the problem persists, please let us know.

closed 2019-02-22 by VinArt