OpenRoberta/robertalab #987: BOB3: IR reflection value

springob, 2018-08-31:
For some examples, we need the refection value from the IR sensor (it worked some time ago - due to a bug, but you fixed it some time ago! ;-) )

existing block / function call:
"gib Umgebungslicht % Infrarotsensor" -> myBob.getIRLight()

please add:
"gib Reflexion % Infrarotsensor" -> myBob.getIRSensor()

perhaps you could just add a drop-down entry "Reflexion"...

VinArt, 2018-10-19:
This needs a new block or drop-down. A note to self: the first option (getIRLight) measures the IR value with the LED turned off, the second option (getIRSensor) performs two measurements, with and without LED backlight and returns the difference.

VinArt, 2019-02-13:
Tested successfully with the following program:

```xml
<?xml version="1.0"?>
<export xmlns="http://de.fhg.iais.roberta.blockly">
  <program>
    <block_set xmlns="http://de.fhg.iais.roberta.blockly" robottype="bob3" xmlversion="2.0" description="" tags="">
      <instance x="384" y="50">
        <block type="robControls_start_ardu" id="#-IH#f(pN,(+qn0NeaPl" intask="true" deletable="false">
          <mutation declare="true"/>
          <field name="DEBUG"/>
          <statement name="ST">
            <block type="robGlobalVariables_declare" id="qXetJ4N=P;)KQml2pZ)f" intask="true" deletable="false" movable="false">
              <mutation next="true" declaration_type="Number"/>
              <field name="VAR">reflectedLight</field>
              <field name="TYPE">Number</field>
              <value name="VALUE">
                <block type="math_number" id="`l6AnFp.[YPEng9f])vZ" intask="true">
                  <field name="NUM">0</field>
                </block>
              </value>
            </block>
            <block type="robGlobalVariables_declare" id="%l87:7eC;jGQ-3fsHzYQ" intask="true" deletable="false" movable="false">
              <mutation next="false" declaration_type="Number"/>
              <field name="VAR">ambientLight</field>
              <field name="TYPE">Number</field>
              <value name="VALUE">
                <block type="math_number" id="vJ9dp=!o@l#|kTr.j.?h" intask="true">
                  <field name="NUM">0</field>
                </block>
              </value>
            </block>
          </statement>
        </block>
        <block type="robControls_loopForever_ardu" id="Q#gk%RPkMmp:co[#y8vA" intask="true" deletable="false" movable="false">
          <statement name="DO">
            <block type="variables_set" id="*xnMuB}[HaZVJJ{%_J1I" intask="true">
              <mutation datatype="Number"/>
              <field name="VAR">ambientLight</field>
              <value name="VALUE">
                <block type="robSensors_infrared_getSample" id=")966?)V*nzPiq5)t]MG|" intask="true">
                  <mutation mode="AMBIENTLIGHT"/>
                  <field name="MODE">AMBIENTLIGHT</field>
                  <field name="SENSORPORT"/>
                  <field name="SLOT"/>
                </block>
              </value>
            </block>
            <block type="variables_set" id="oLeNW78ZKmi(|},.8_ak" intask="true">
              <mutation datatype="Number"/>
              <field name="VAR">reflectedLight</field>
              <value name="VALUE">
                <block type="robSensors_infrared_getSample" id="]NZWR.KKQin`{52hy+XS" intask="true">
                  <mutation mode="REFLEXION"/>
                  <field name="MODE">REFLEXION</field>
                  <field name="SENSORPORT"/>
                  <field name="SLOT"/>
                </block>
              </value>
            </block>
            <block type="robControls_ifElse" id="Rw2^ICdp6VU47I~W|2Ar" intask="true">
              <mutation else="1"/>
              <repetitions>
                <value name="IF0">
                  <block type="logic_compare" id="ERa1H=Jv]VNbvHBP|@|c" intask="true">
                    <field name="OP">LTE</field>
                    <value name="A">
                      <block type="variables_get" id="qI.}mmaoYGh8hMgw_F5Q" intask="true">
                        <mutation datatype="Number"/>
                        <field name="VAR">reflectedLight</field>
[... 37 lines cut ...]
                  <block type="makeblockActions_leds_on" id="jjTq},hZyzNz)nF}mb6!" intask="true">
                    <field name="LEDSIDE">Right</field>
                    <value name="COLOR">
                      <block type="robColour_picker" id="5)SKcN:rN:BH]b!PS3JG" intask="true">
                        <field name="COLOUR">#FF0000</field>
                      </block>
                    </value>
                  </block>
                </statement>
              </repetitions>
            </block>
          </statement>
        </block>
      </instance>
    </block_set>
  </program>
  <config>
    <block_set xmlns="http://de.fhg.iais.roberta.blockly" robottype="bob3" xmlversion="2.0" description="" tags="">
      <instance x="213" y="213">
        <block type="robControls_loopForever_ardu" id="1" intask="true" deletable="false" movable="false"/>
      </instance>
    </block_set>
  </config>
</export>
```

closed 2019-02-13 by VinArt