OpenRoberta/robertalab #993: no clear design decision w.r.t. to missing values

rbudde, 2018-09-05:
if values are missing (example: Calliope), defaults are used ("" or 0 e.g.). Sometimes NULL is used (raises sometimes warnings from the cross-compiler). The program compiles.

My opinion here is: it should be checked, a (good visible!) error attached to the blocks sent back to the user and the compile should be suppressed. I think, these defaults are especially bad from a didactic point of view.

The semantics of NEPO-programs should be documented in the wiki.

VinArt, 2018-09-06:
[image]

This program does not compile, since some values that are left empty produce nothing and not a default value. Variables that are not set are not initialised as well.

VinArt, 2018-09-13:
[image]
Default value for connection is missing for the send message block as well as for the receive block:
[image]

VinArt, 2018-09-21:
A possible solution that was discussed in today's meeting: enhance the block checker and when the values for blocks are empty, provide an error (like observed for the "and" block) so users would complete their programs.

It is still unclear what to show in the code pane to the user though.

closed 2019-06-12 by bjost2s