OpenRoberta/robertalab #994: Calliope and add a value to a list

ToniTaste, 2018-09-05:
Dear OpenRoberta Team,

there seems to be a bug in adding values in lists.
I attach an example where the bug gets obvious.
[CalliopeList.zip](https://github.com/OpenRoberta/robertalab/files/2354008/CalliopeList.zip)

In the simulator everything works fine. But in the Sourcecode there are two semicolon at the end of the line. I think thats the reason why programm on the Calliope mini crashes.

VinArt, 2018-09-06:
Hi @ToniTaste I think it is actually a bug in the simulation. I have loaded the XML, you've sent to produce the following picture:
[image]
What we see here is a creation of an empty list and then execution of "set" operation on an non-existent element (since list is empty) this should result in a segfault, since we are trying to access memory out of bounds.

If we initialise the list with a one with 1 element, or use a "Number" type variable, the segfault will be gone.

ToniTaste, 2018-09-06:
Well, I think there are two problems: 
1. When I use the lists block 'in list ... set ...' then two semikolons will be display in the source code window.
[image]

2. It is not clear to my how I add a new value to an empty list. It is not possibility by using this one:
[image] 
Therefore it works in the simulation (and not on the calliope mini) by using this one:
[image]
Or is a list not a dynamic data structure but only an array with a fixed size.

rbudde, 2018-09-07:
related to #797 

The question above is ok: we call it list, but it is a fixed-size array without index-out-of-bounds-check :-). This has to be changed.

At the moment the only solution is: create an array of size 40 (not too much, as the memory in Calliope is very limited), for instance, and put the data into it

boonto, 2018-09-07:
This issue should be solved when the concept devised is #797 is implemented.

VinArt, 2018-11-22:
_setListElementByIndex is used currently when generating source for the above program, since std::list is used from now, this should not be an issue anymore.

closed 2019-03-23 by bjost2s