OpenRoberta/robertalab #1006: c++ code generator: expressions in math not enough parenthesized

rbudde, 2018-09-24:
**Describe the bug**
 - c++ code generator: expressions in math not enough parenthesized

**To Reproduce**
1. use calliope
2. create the program "show text(sin(1+2))"
3. inspect the code generated: "_uBit.display.scroll(ManagedString(sin(PI / 180.0 * 1 + 2)));"
4. not what the user wantwed :-)

**Expected behavior**
 - code generated should be "(1+2)" instead of "1+2"

**Definition of Done**
1. error corrected
2. creating a new issue: write test programs to guarantee correct behavior for all robots

kunkunuzzi, 2018-09-29:
Seems related to #983 and  #972.
Maybe all math operators (and similiar blocks like #972) should simply have always an inner a and an outer bracket:
e.g. translate the add block to:
(  ( <field1> ) + ( <field2> )  )
So all 'user code' inside and outside is safely encapsulated by brackets. And extra brackets don't hurt.

VinArt, 2018-11-12:
I actually fixed this for NXT, for binary expressions, where the operation is arithmetical. I should probably pull this up for all c++ codegeneration

rbudde, 2018-11-29:
fixed for all c++ based robots in AbstractCppVisitor and for nxc in class NxtNxcVisitor

closed 2018-12-20 by VinArt