OpenRoberta/robertalab #1007: Calliope support for serial read / write

bjost2s, 2018-09-25:
**Is your feature request related to a problem? Please describe.**
 - As a calliope programmer you run often into problems when dealing with sensor values. Because there is no way to debug calliope so far, a serial write function would make it much easier to debug the program. 

**Describe the solution you'd like**
 - here is a good description of the feature https://mini.pxt.io/device/serial

VinArt, 2018-09-26:
Microbit allows the usage of serial as well, if one would connect to the serial device created by microbit, the python console would be shown.

boonto, 2018-11-20:
I started to implement the serial write for Calliope.
I moved the Arduino specific write serial up to the common robot and added the block to the Calliope expert *display* toolbox.
It works for now, however I am not sure what mode is best (`ASYNC`, `SYNC_SPINWAIT` or `SYNC_SLEEP`; see [MicroBitSerial.cpp](https://github.com/OpenRoberta/microbit-dal/blob/master/source/drivers/MicroBitSerial.cpp#L424)) and there is still some weird behaviour in combination with `wait` blocks.

Still missing:
- [ ] Write serial implementation in Microbit visitor
- [ ] Write serial implementation in simulation visitor
- [ ] Adding read serial block to mbed toolbox
- [ ] Read serial implementation in Calliope visitor 
- [ ] Read serial implementation in Microbit visitor 
- [ ] Read serial implementation in simulation visitor 
- [ ] Tests

VinArt, 2018-11-26:
Only serial write action is wanted

rbudde, 2019-02-12:
tested, does work. Added the remaining (small?) tasks to #1070

closed 2019-02-12 by rbudde