OpenRoberta/robertalab #1008: EV3Sim Lists wrong behaviour

boonto, 2018-09-27:
**Describe the bug**
There are two wrong behaviours in the List functions for the EV3 simulation.
1. `in list x insert at # from end` the inserted element is misplaced
2. `in list x remove #` does not delete the element

**To Reproduce**
1. Create the following program:
[image]

2. Create the following program:
[image]

**Expected behavior**
1. The shown list is [0, 0, 0, 1]
2. The shown list is [0, 2]

VinArt, 2018-11-22:
1. actually this is indeed the expected behaviour, because the insert operation puts an element at the desired index and if any element exists prior to this operation at this index it (and all the elements to the right) will have their index incremented by one
2. needs checking, probably the operation cannot be done in-place.

VinArt, 2018-11-22:
Fixed 2.
[image]

closed 2019-03-23 by bjost2s